NikodemBartnik/IndyMill #6: Full 3D model?

eyesonlybob, 2021-12-29:
Awesome project! I don't see a full 3D model available. Can you add one? I'm interested in building one but want to inspect the assembly closer first.

Kosh42, 2022-01-02:
What I was after too. To be fully Open Source, the source should also be available to inspect, and adapt. Otherwise free to build is possibly more accurate description.

Still an awesome project. I just want to scale it down for my needs.

NikodemBartnik, 2022-01-02:
You can find all the STEP files for 3D printed parts and DXF for the plates, based on that you can create your own assembly that will suit your needs and modify whatever you want. That is the source of the project

closed 2022-01-02 by NikodemBartnik

eyesonlybob, 2022-01-02:
That just makes things more difficult than it needs to be.  You're limiting
further development of indymill by not releasing the full model.