NikodemBartnik/IndyMill #13: Spindle PWM  and Light PCBs - can't open them.

FSXLiveriesdotcom, 2023-04-02:
Hi @NikodemBartnik
I have problems with opening SCH and BRD files. 
I tried to open SCH direcly in Eagle (and directly import in Fusion eCAD) but got error:

 **_Loading Y:/Downloads/spindle_light.sch ...
Error:
line 9, column 129: This is not an EAGLE file._**

Same error happens when trying PWM controller files.
EasyEDA and KiCAD also can't import them.

Plus I have additional questions as some things are confusing. I must admit,  my understanding of electronics here might be a issue here.

I uploaded controller gerber files to JLCPBCB to at least see 3D design and try to recreate the PCB from scratch.
1. It seems that apart from resistors and capacitor, there is only one other component - amplifier LM358DR
   However, in BOM file there is also mentioned voltage regulator which doesn't have any placement on PCB. 
   Am I missing something? 
2. Also,  could you make some simple drawing to show up how to connect spindle > PCB > PSU? 
   Similar pic to one you have done for stepper motors and drivers. That would be very useful.

3. Why there are pin additional pin headers on PWM PCB?  What is their purpose?

Greetings, Mark

NikodemBartnik, 2023-04-03:
All the files open perfectly fine for me (I am using Eagle 9.5.1).
The voltage regulator is an old part that in the end wasn't used in the project.
The pin headers are there in case you prefer to connect it with breadboard cables instead of screw terminals

FSXLiveriesdotcom, 2023-04-20:
Hey Nikodem. 
OK, I tried once more and was able to open the files but only after downloading .zip of the entire repo. 
When tried first time I was downloading single files only.

closed 2023-04-20 by FSXLiveriesdotcom